Tympan/Tympan_Rev_E_Hardware #2: Increase capacity of the 3.3V power supply

chipaudette, 2020-05-21:
In my experience running 4 channels at high speed with SD writing, the system becomes unstable.  I believe that the power draw is too large for the existing regulator, given the voltage of the LiPo battery.

Recent current draw measurements suggest that we can pull 200-300 mA, which is nearing the max of the current [AP2210 regulator](https://www.diodes.com/assets/Datasheets/AP2210.pdf).  Moving to the Teensy 4, where we can move to higher CPU speeds, will likely increase this power consumption even more.  I think that we should consider bigger power supplies.

Looking at digikey, the nearest bigger regulator is the [AP2213, 500 mA](https://www.diodes.com/assets/Datasheets/AP2213.pdf).  What I see, however, is that the total capacity may be higher, but that the voltage drop is the same for a given current draw.  So, if our problem is that our battery voltage just isn't high enough for the regulator to deliver the required current, switching to the 2213 might not actually solve the problem.

Does anyone know of a low-noise regulator that has even less voltage dropout in the 200-400 mA range?

biomurph, 2020-05-21:
I just did a quick look at Adafruit WiFi module ([HUZZAH ESP32](https://learn.adafruit.com/adafruit-huzzah32-esp32-feather/downloads?preview_token=0X3pzGCh9fm47-K5awpMrA) as those things are usually super power hungry and Adafruit is good at power supply design. They use the [AP2112](https://www.diodes.com/assets/Datasheets/AP2112.pdf), which has a high output current (600mA) and what seems like darn good drop out voltage. Not enough time right now to do a head-to-head comparison, but worth a look.

eyuan-creare, 2020-05-22:
Good thought Joel.  Never thought of looking at what other products use.  Here's a nice LDO, but it's expensive ($6).  Two package sizes:
[ADP7156ACPZ-3.3-R7](https://www.digikey.com/product-detail/en/analog-devices-inc/ADP7156ACPZ-3-3-R7/ADP7156ACPZ-3-3-R7CT-ND/5957834)
[ADP7156ARDZ-3.3-R7](https://www.digikey.com/product-detail/en/analog-devices-inc/ADP7156ARDZ-3-3-R7/ADP7156ARDZ-3-3-R7CT-ND/5957841)

Specs
* Dropout: 50-120mV
[image]

* PSSR: -80dB
[image]

eyuan-creare, 2020-05-22:
Noting the PSSR for the existing AP2210 regulator:
[image]

eyuan-creare, 2020-05-22:
If we are happy with 0.29V dropout, then there are some for $0.50 with 700mA output and low PSSR.
[NCP167 series](https://www.digikey.com/products/en/integrated-circuits-ics/pmic-voltage-regulators-linear/699?k=ldo+regulator&k=&pkeyword=ldo+regulator&sv=0&pv2065=17895&sf=0&FV=69%7C409393%2C129%7C119750%2C129%7C167594%2C129%7C201594%2C129%7C255853%2C129%7C272951%2C129%7C277416%2C129%7C286223%2C129%7C290791%2C129%7C45559%2C129%7C47316%2C129%7C48377%2C129%7C50900%2C129%7C55041%2C129%7C60103%2Cii1%7C1065%2C1779%7C174115%2C1779%7C174120%2C1779%7C174258%2C1779%7C174288%2C1779%7C174290%2C1779%7C174321%2C1779%7C174378%2C1779%7C310372%2C2064%7C287128%2C2064%7C287190%2C2064%7C289580%2C2064%7C291246%2C2064%7C291250%2C2064%7C302557%2C2065%7C13044%2C2065%7C17895%2C-8%7C699&quantity=&ColumnSort=1000011&page=1&stock=1&pageSize=500)

eyuan-creare, 2020-05-22:
AP2112 has a 0.25V dropout, which is nice.  PSSR is around 65dB:
[image]

And AP2213 only defines PSSR for 10mA, but it is around 60-70dB
[image]

 It has this note, which suggests that for good PSRR, Vin should be 1V higher than Vout.  Likely true for other LDOs as well.  
[image]

eyuan-creare, 2020-05-22:
Nice reference on choosing LDO regulator specs from [Analog](https://www.analog.com/en/analog-dialogue/articles/understand-ldo-concepts.html)

eyuan-creare, 2020-05-26:
Alternatively, we could spec a buck boost converter for providing 5V to the BT module.  That would keep it from sagging the 3.3V bus when it sporadically draws current.

chipaudette, 2020-06-17:
Nice work, team!

For the AP2112, we have a working schematic from Adafruit and we have Adafruit's seal-of-approval.  That seals the deal for me.  Let's go with that.

chipaudette, 2020-07-06:
Adding a Buck-Boost option (as suggested earlier), for completeness.  Shouldn't be needed with a LiPo to support by 3.3V, but there is a certain appeal:

TPS6303x:  https://www.ti.com/lit/ds/symlink/tps63031.pdf

Let's stick with the AP2112 because of the adafruit documentation.

Chip

biomurph, 2020-08-03:
Moving to the Teensy 4.1 and reviving this topic.
The schematic for the 4.1 is [HERE](https://www.pjrc.com/teensy/schematic.html)
Paul is using the [TLV75733](https://www.ti.com/store/ti/en/p/product/?p=TLV75733PDRVR) Vreg.
The output current limit seems to be adequate. @eyuan-creare is the PSRR and drop out OK?

erk1313, 2020-08-03:
@Chip: @biomurph suggested the TLV75733, which is part of the Teensy 4.1 design.  It has a high output current (1A), a low dropout, and a decent PSSR.  The PSSR is less than our existing Rev-D regulator, but that's probably not important as we are not feeding it from a high-switching supply, nor have components that create high frequency noise.  I would say it's fine.  

| IC | Max Current | Dropout @500mA | PSR @ 10kHz; 500mA | PSSR Plot |Link |
|----|----|----|----|----|----|
| AP2112 | 600mA | 250mV | 67dB |[image]|[datasheet](https://www.diodes.com/assets/Datasheets/AP2112.pdf)|
| TLV75733 | 1000mA | 175mV| 45dB | [image]|[datasheet](https://www.ti.com/lit/ds/symlink/tlv757p.pdf?ts=1596461065952&ref_url=https%253A%252F%252Fwww.ti.com%252Fstore%252Fti%252Fen%252Fp%252Fproduct%252F%253Fp%253DTLV75733PDRVR)

chipaudette, 2020-08-12:
Sounds good...let's go with the TLV75733.

closed 2020-09-25 by chipaudette