Tympan/Tympan_Rev_E_Hardware #3: Design for T4.0 or T4.1?

chipaudette, 2020-06-17:
I believe that you had found solutions for the all those difficult-to-use pads on the T4.0, but why should we stick with the T4.0 form-factor when we could just switch to the T4.1 and use much-friendlier female 0.1" headers?

biomurph, 2020-06-18:
ooooooooh yeah!!
Life just got easier.

Looking at the specs, it is a nice beast! USB Host! Ethernet! Memory expansion! 

I'm gonna get mine today!

biomurph, 2020-09-03:
@chipaudette I have the board laid out and routed. 
It looks good to me, and I'd like to green light a prototype run. 
Please take a look at the work and let me know what you think.

I have the Teensy mounted on the bottom of the board. Can do with mating headers, or can do by soldering SMT pins into the teensy 4.1 for low profile.

Same form factor as the Tympan Rev D, with breakouts, board size, mounting holes, volume knob, power switch, LEDs all in the same position (except for the charging LED...)
I plan to route a slot on the right of the board to allow the battery cable to snake to the back of the board.

biomurph, 2020-09-08:
@chipaudette 
The board is ready to go! Please take a look at the schematics and let me know if you see anything that needs attention, otherwise I am prepared to green light this design and get the BOM ordered.

closed 2021-06-10 by chipaudette