Tympan/Tympan_Rev_E_Hardware #4: Startup of BT Module

chipaudette, 2020-06-17:
@eyuan-creare and @biomurph , 

On RevD, we have the issue where sometimes the BT module doesn't start when the rest of the system starts.  The most obvious symptom is that the 3-color LED attached to the BT module never comes on. 

We have talked about this.  But did we ever identify the specific circuit changes that we wanted to implement to try to make this problem better?  If so, we could try some here on Rev E.

Does anyone have any notes?

eyuan-creare, 2020-06-17:
Captured with the Rev-D issue: https://github.com/Tympan/Tympan_Rev_D_Hardware/issues/2 
The BT Module datasheet provides an example circuit, I believe.  If not, let me know and I can dig up a blog post that went into more detail.

biomurph, 2020-06-18:
That RC circuit referenced is very easy to add.

A (maybe related) issue is one I've noticed, and would appreciate if someone could try to replicate it.
I'm attempting to turn the BC127 on and off by controlling the BT_nRESET (pin 34).
I'm using a pushbutton, but I'm sure that it could be toggled with a serial prompt.
The issue is that the BC127 does not always come out of reset when I try to turn it on. Maybe I'm holding the reset pin low for too long... It does wake up when I only reset for a short while (seconds) but if long(er) then it's no-go. 
Maybe I'm doing it wrong, but my goal is to turn off the BT module when not in use, so that it will save power.

eyuan-creare, 2020-06-19:
Transferred the Rev-D issue to this repo.  Closing this so it is in one spot.  Refer to #5

closed 2020-06-19 by eyuan-creare