Tympan/Tympan_Rev_E_Hardware #6: Add ability to disconnect the battery for shipping

eyuan-creare, 2020-06-29:
Transportation

eyuan-creare, 2020-06-29:
* If the battery was a button-top, then you could insert a small strip of plastic as the disconnect.
[image]

eyuan-creare, 2020-06-29:
@chipaudette We could either replace the battery charger IC or add a load switch with ship mode.  Ideally, the design would not require a momentary on/off button.

If you like these option, I can mock up a design to pass by Ray.  I like the fact that there are reference designs we can point to.

biomurph, 2020-08-03:
@eyuan-creare I"m not sure if we need this kind of feature. We ship with a low capacity LiPo (500mAh) and I have found that if you are shipping them contained in equipment. Check out the end of page 4 on this [doc](https://www.ups.com/assets/resources/media/en_US/pack_ship_batteries.pdf). I don't think that we need to worry about Tympan due to the small battery size. Once you get into bigger battery capacity, that's when there may be question, but it seems the cutoff is pretty clear.

(unsure why I'm just seeing this, will check my settings)

erk1313, 2020-08-03:
@biomurph There's a whole range of regulations for shipping Li-ion batteries.  I have a summary I can share at some later point. 
 If it weren't for the fact that we qualify as a low-production prototype, we would require UN38.1 testing for the completed assembly.  I'll be reviewing how Creare interprets these regulations with Creare's safety person in August and will check back then.  Their initial ruling was that a physical disconnect was required, but I think that is too conservative a ruling.

biomurph, 2020-08-03:
Sounds good, @eyuan-creare 
My research has only gone as deep as UPS and DHL.

eyuan-creare, 2020-09-08:
After a review by Creare's safety team, the Rev-D design was deemed safe for shipping with the battery connected, with these three design changes:
* Place the fuse immediately after the battery
* Move the power switch before the voltage regulators

Notes contained on networked folder: \\OLYMPUS\Projects\7146-Cool-Aid\Technical Work\Design\2020-0710_Shipping Lithium Batteries

[image]

biomurph, 2020-09-08:
@erk1313 
I can make those changes to the Rev E prototype design.

chipaudette, 2020-09-25:
@biomurph 
Was this change fully implemented in Rev E?  

Looking at the current RevE schem, I think that I see that the switch has been moved, but I don't think that the fuze is as early in the circuit as is shown here in the figure above.  It's immediately after the battery in the figure above, but it appears to be after the switch in the current RevE schem...

biomurph, 2020-09-29:
@chipaudette Changes done on my local. Will push to repo.

biomurph, 2020-09-29:
Just for the sake of clarity on this. The resoning we had in the beginning to arrange the power supply this way was to make sure that the RTC [V-BAT] was powered by the 3.3V. We gave users the option to add their own external power for the V-BAT. In order to do this, when the Tympan D was 'off', there was power going to the RTC only, so the 3.3V reg was running to do that when the switch was 'off'. In this new version, we are adding a coin cell to power the RTC, so we don't need to run the LiPo to do that work. That's why in previous the switch was after the Vreg.

closed 2020-09-29 by biomurph