Tympan/Tympan_Rev_E_Hardware #7: Fuel Gauge for Battery Level

eyuan-creare, 2020-09-08:
Add a fuel gauge (or spec a charging IC with that feature) to report the estimated battery level of the device.  This will allow the Tympan to close an SD recording before a power failure.  Note that Lithium batteries cannot rely on voltage to determine charge level.  

@chipaudette : is this a nice-to-have feature or a must?

biomurph, 2020-09-09:
Here is a candidate for fuel gauge
https://www.mouser.com/datasheet/2/308/LC709203F_D-1810548.pdf

eyuan-creare, 2020-09-09:
Thanks for checking into these.  I've never used one of these chips before.  I see that it uses a special technique that doesn't use a sense resistor, but it looks like it relies on a thermistor.  Are there other reliable methods of measuring battery capacity?  I simply ask because it adds another component to the battery.
[image]

chipaudette, 2020-09-09:
Did we really talk about doing this for Rev E, or is it for the MIT hardware?

chipaudette, 2020-09-09:
@biomurph, that's the same IC as used in the recently-advertised Adafruit board, right?

https://www.adafruit.com/product/4712

eyuan-creare, 2020-09-09:
Looks like Adafruit is using it without a thermistor, so they forgo temperature compensation.  

[image]

@chipaudette Yes, we talked about it in the context of the Dosimeter.  We can shelve this if not needed.

biomurph, 2020-09-09:
@chipaudette Yes! it does look like Adafruit is using the same IC!
Funny, I did not even check to see if there was one broken out by the SparkFruit people. 

@eyuan-creare I selected the LC709203 because it does not require an external current sense resistor, and also noticed that it can be used with or without thermistor temperature measurement. 
It seems to be the easiest to implement, and since we are not doing anything that is high current, or really stressing the battery in any way, I thought that the temperature measurement would be optional. Do you guys feel that it is something that we should look into?

If you're interested in playing around with this IC in the Tympan Library, I can add it into the Tympan Rev D spec and we can drive it around a bit?

chipaudette, 2020-09-25:
I'm interested in this, but I'm more interested in getting Rev E built for the first time.  So, let's not do this (yet) for Rev E.

biomurph, 2020-09-29:
@chipaudette There is room for it, and I can route it in but leave it unpopulated for now. How does that sound?

biomurph, 2020-09-29:
I added it into the layout, and we can decide to populate it or not. Super easy.

closed 2020-09-29 by biomurph