Tympan/Tympan_Rev_E_Hardware #10: Bring Teensy USB signal up to Tympan (for power and programming)

eyuan-creare, 2021-01-08:
To expose the SD card slot in the enclosure, one _possibility_ is sliding the Teensy forward. However this would hide the micro USB port.  Depending on complexity, we may want to bring up the USB signal to the Tympan to relocate a port there.  On the plus side, this would allow us to use a USB-C connector.

Here’s the pins labeled D- D+ on the side opposite of the Teensy 4.1.  I outlined the traces on that side of the board.
  
[image]

I flipped that trace to match it up with the other side.  That should correspond to the fingers on the micro USB connectors: D- (2) and D+ (3).  I'll pint this out when I get my hands on a Teensy 4.1 today.

[image]

@biomurph, @chipaudette  Got any ideas for how we could bring the USB signal up?  Looks like it is 0.1" spacing.  Would a header fit these pads?

eyuan-creare, 2021-01-14:
I tried soldering a header (0.1in pitch) to the USB D+ D- pads, but failed and lifted up a pad.  If the Teensy is first soldered to the Tympan, then perhaps there's low risk of lifting the pad.

@biomurph What are your thoughts?  Do you think soldering a header to these pads is feasible, or would you suggest wire leads?  
- We could also do pogo pins (if their height was short).  They would likely be fine under vibration, as their spring force can be rather high.  But there is a possibility of corrosion over time.

biomurph, 2021-01-14:
It looks like the solder pads are 0.1" spacing? 
If so, that makes it easier to cope with. 
Then the issue becomes finding room to mount the USB C...

What about using Occam's USB converter?
https://www.wish.com/product/5e1e787f93907f0566eb2d2d?hide_login_modal=true&from_ad=goog_shopping&_display_country_code=US&_force_currency_code=USD&pid=googleadwords_int&c=%7BcampaignId%7D&ad_cid=5e1e787f93907f0566eb2d2d&ad_cc=US&ad_curr=USD&ad_price=1.00&campaign_id=7203534630&gclid=Cj0KCQiA9P__BRC0ARIsAEZ6iriuBtoj-72D2dnFmsrzpXKMggnzykg-SZafDkFJtI2EJGfBTrLgrLwaAjgtEALw_wcB&share=web
Might this be long enough to make the distance?

eyuan-creare, 2021-01-14:
@biomurph The adapter is a great idea and converts it to USB-C.   I'm a little sketched out by using an extension, but that may be less sketchy than adding headers/pogo pin/leads to the USB pads.  They are 0.1" spacing.  Would you like to try experimenting on one of the Teensy's you have designated for us?  Even if you lift the pad, it shouldn't effect the USB connection.  I think that pad is just a test point for them to program the device (just a guess).

I'll order that Amazon adapter in and take a look at how rigidly it is held.

biomurph, 2021-01-14:
@eyuan-creare I will take a look at it. I am hesitant to use pogo pins, as they present their own mechanical issues, but they may be the best answer if we do our own connector. I can see that it will be difficult to get the solder points done in situ with the boards so close together and the USB C mounted likely right near there, so I would think that the pogos, if they are available at the right length, might be our best bet.

I'm looking at the layout and there seems to be room to put SMT pads or maybe through holes in the area where they might need to land.

Another issue with 'sliding the teensy' is that the pads for the Teensy may be crashing against the audio jack mechanical mount holes??

[image]

eyuan-creare, 2021-04-12:
The USB extension looks a little sketchy.  Too much leverage to pop the USB connector off.<br>
[image]

eyuan-creare, 2021-04-14:
@biomurph: Chip thinks the risks of either approach outweigh the benefits of exposing the SD card.  Let's abort this request.

closed 2021-04-14 by eyuan-creare