Tympan/Tympan_Rev_E_Hardware #9: Bring Teensy reset button up to the Tympan

eyuan-creare, 2021-01-08:
In the existing enclosure, the battery obscures the reset button.  If we brought the reset signal up to the Tympan, then we could place a second reset button in a better location.  Likely, this would be a "side-press" button, but that is TBD.

- Existing enclosure:

[image]

- Teensy 4.1 Schematic showing the PROGO trace from the reset button:
[image]

- Teensy 4.1 PCB showing the trace from the PROGO pin to the Reset Button.  @chipaudette confirmed that this thru hole pin transitions LOW when the reset button is pressed

[image]

eyuan-creare, 2021-01-08:
@biomurph looked into where the pin would fall:

Here is about the location of the pad that we would need to place for it.
[image]

 

The good news is that the pad for the pin next to it (pin 30) is not in the way.
Also, there's enough room to either place a thru-hole, with a bunch of parts/trace juggling, or placement of a single SMT pad, with less parts/trace juggling.

There's also the option of flying a lead from the T 4.1 to the Tympan..

eyuan-creare, 2021-05-17:
@biomurph suggested placing a momentary pushbutton (TL6340AF160Q) between the audio jacks:

[image]

This should work fine, though it is a tiny switch.  The button is 1/32" tall. @chipaudette if you fine using a paper clip, I can put a guide hole in.  If you would like a feature on the case with a cantilevered tab, then we should go bigger.

biomurph, 2021-05-17:
possible candidate switches
https://www.mouser.com/datasheet/2/910/909_2894_2893_DTSA-6-1571568.pdf
The DTSA-6 drawing will fit and it is a decent size. There are tons of them.

eyuan-creare, 2021-05-17:
Goldilocks says... that switch is just too big!  It does fit, within the height of the stock headers but let me see it there is an alternative.

[image]

eyuan-creare, 2021-05-17:
@biomurph What do you think of this switch? (faked up CAD model).  It still feels a bit big but there is not much out there for RA switches. 
- APEM MJTP1117 ($.48 [Digikey](https://www.digikey.com/en/products/detail/apem-inc/MJTP1117/1795496)) SPST- NO
<br>
[image]
<br>
[image]

[image]

biomurph, 2021-05-17:
Works great! Im down to use it. 
You have it overhanging the edge of the PCB. Is that how you want to?

eyuan-creare, 2021-05-17:
@biomurph yes, please have the button overhang. Thanks!

biomurph, 2021-05-21:
I placed the MJTP-117 push button between the audio jacks

chipaudette, 2021-06-10:
@eyuan-creare , is this done?  can we close this?

closed 2021-06-10 by eyuan-creare