Tympan/Tympan_Rev_E_Hardware #12: Add Thru Holes for Reuse of GPIO lines to RGB LED

eyuan-creare, 2021-05-17:
@biomurph : *if there is space*, it would be nice to break out the GPIO going to the RGB LED.  I'm thinking (3) thru holes spaced 0.1in apart. But only if you have space.

[image]

biomurph, 2021-05-17:
I will see. What is the reason for this breakout?

eyuan-creare, 2021-05-17:
For additional I/O.  It's not critical but if you have space for it, we've run out of GPIO in the past when we are using up the available pins for I2C, PDM, etc...

biomurph, 2021-05-17:
Ah, those pins come from the BC127. If you want more Teensy pins, that's a different thing

eyuan-creare, 2021-05-18:
Ah.  You're right.  I'm glad you asked about the why.  Well, if there is room to put 2-pin thru holes next to LED_A & LED_R, that would work as well.  And if not, we'll close this.

[image]

biomurph, 2021-05-18:
I put in two breakout GPIOs on the 2mm header at the bottom left.

Pin 34 and 35 are also UART 8 TX and RX

[image]

eyuan-creare, 2021-05-18:
Thanks for correcting me.  For some reason, thought we were using 0.1in spacing.

biomurph, 2021-05-18:
Well, we are using 0.1" for the breakout headers that the shield plugs into. the 2mm header was what would fit to make the RTC battery power on Rev D. That 2mm header row has now grown a few pins.

eyuan-creare, 2021-05-18:
Ah.. now I understand.