Tympan/Tympan_Rev_E_Hardware #14: Room for QUIIC?

chipaudette, 2021-05-18:
@biomurph There's no room for a QUIIC connector anywhere is there?  (I do love QUIIC)

biomurph, 2021-05-19:
It looks like a Qiic connector could fit under the volume knob... lousy location...
Mayby I could throw one in above MIC1 next to the speaker jack?

biomurph, 2021-05-19:
Here's a shot of the Qwiic location that is the easiest to do.
It's at the front of the board, on the bottom layer, underneath the PROGRAM pushbutton between the audio jacks.

[image]

biomurph, 2021-05-21:
Placec Qwiic connector on the bottom near the battery connector. It fits!

[image]

closed 2021-06-10 by chipaudette