Tympan/Tympan_Rev_E_Hardware #15: Decide Direction for the Rev-E Enclosure

eyuan-creare, 2021-06-09:
- [ ] Determine what the stackup height is.  @biomurph is planning to solder the Teensy to the Tympan

biomurph, 2021-06-10:
@eyuan-creare do we need the belt clip? Do you have a part number for it?
I'm thinking it would be loose in the bag for users to apply if they want. So it could still rest on the table-top.

eyuan-creare, 2021-06-10:
@biomurph it's a bit pricey for what it is: [BELTCLIP MCS-661BP](https://www.theclip.com/store/metal-belt-clip-661).

I'll upload the CAD for the triple stack design that has the whole assembly.  What SW version are you working with?

biomurph, 2021-06-10:
Here's one that is a bit cheaper.
[belt clip](https://www.amazon.com/PSCCO-552510mm-Holster-Sheath-Spring/dp/B08NWSWCXG/ref=sr_1_13?dchild=1&keywords=belt+clip&qid=1623332485&sr=8-13) but it has a different shape for the mount surface...

I use FreeCAD. I don't do much with it. I'm not a versatile CAD user. 
I also use TinkerCAD, which I find useful for making small changes and simple forms.

eyuan-creare, 2021-06-11:
@biomurph I like the idea of designing it so that anyone can print it with PLA.  Is there a UV protecting finish you could apply?

biomurph, 2021-06-11:
I will look into uv safe filament. 
Would be better not to do a coating. That sounds messy.

Yes, the wan Shun models are good. I reworked them in tinkercad to make them print better on my machine

eyuan-creare, 2021-06-11:
Oooo..  Can you share any photos of your printed earpieces?  What wall thickness can you do?

biomurph, 2021-06-11:
Oh, I'm sorry I misunderstood.
I'm not talking about the earpieces I'm talking about the Rev D enclosure.

I have not printed the earpieces. Not worth it, since I figure the resolution will not come out well enough to use.

eyuan-creare, 2021-06-11:
- Go with (7) design from the SD Card issue #18 using Adafruit 2000mAh Li-poly battery
- Consider sliding case and front-screwed bottom
- Consider leaving extra height for battery (so no accidental reset button presses) and accommodate other batteries
- Leave holes for PCB mic (with funnel shape), LEDS (GPIO, BT, Charging)
- Consider "ASA Special Edition" embossing
- Note that Joel's printer can do 0.080" down to 0.040".  Joel to look into UV stable filament
- Aim for cases to be complete before June 21
- New changes to PCB include the reset button and Quic connector

biomurph, 2021-06-12:
Here's a shot of a prototype Dosimeter case print. Think it was done with a fast setting (lower resolution). I'm going to find a similar blue to the one you have 
[image]

biomurph, 2021-06-12:
Here's the blue that I'm getting to work the prototypes and first cases with
https://www.matterhackers.com/store/l/pro-series-pla/sk/MFEFY96W

eyuan-creare, 2021-06-16:
### Change to Tympan Rev-E.2 PCB that affect enclosure
- Reset button added to **bottom** of Tympan between mic jacks
- Added Quiic connector 
- @biomurph The model exported from KiCAD shows male headers.  Is this correct? Or are we using the same headers as the Tympan Rev-D?
[image]

biomurph, 2021-06-16:
Your changes should reflect that 

1. The Rest Button is added to the TOP of the Tympan between the mic jacks
2. Added quick connector on the BOTTOM near the rout for the battery wire
3. That 3 pin header connector is actually the dial pot 

Please see photos below from Circuit Hub

[Final Assembly Top](https://circuithub-static1.s3.amazonaws.com/orders/9547/attachments/lNMwFSlRScOHXfjlztSQ_IMG_20210615_201836.jpg?Expires=1623871800&AWSAccessKeyId=AKIAYP6GSUAC6FLCU7TC&SignatureMethod=HmacSHA256&Signature=62jFtfkNEskQP6GvlhcdrAlPoZA%3D)

[Final Assembly Bottom](https://circuithub-static1.s3.amazonaws.com/orders/9547/attachments/zcIaBg9SregnpMCFA7E1_IMG_20210615_201904.jpg?Expires=1623871800&AWSAccessKeyId=AKIAYP6GSUAC6FLCU7TC&SignatureMethod=HmacSHA256&Signature=2%2B%2BZ2GBhASOWFq%2F3Lpzp6iVru8s%3D)

eyuan-creare, 2021-06-16:
@biomurph What is the standoff between the Tympan and Teensy?  I show a gap of 0.150in.

[image]

eyuan-creare, 2021-06-16:
Also, which part are you using for the Quiic connector?

[image]

[image]

biomurph, 2021-06-16:
I think that the Qwiic connector might have miss hit the BOM in our repo.
It is listed in the [Circuit Hub Part List](https://circuithub.com/projects/biomurph/Tympan_Rev_E/revisions/35488/parts)
in this case, it is a [SparkFun](https://www.mouser.com/ProductDetail/SparkFun/PRT-16766?qs=TiOZkKH1s2S5E8Ch5jqJTg%3D%3D) part sold by Mouser.

I can confirm your measurement of 0.155" +/- 0.008" tolerance is enough wiggle room.

eyuan-creare, 2021-06-17:
Thanks.  I was able to source CAD for those two parts.  The Sparkfun Quiic connector is actually JST BM04B-SRSS-TB.

@biomurph Can you take a quick look to confirm I have all the important components on board?<br>
[image]

[image]

biomurph, 2021-06-17:
Stand by

biomurph, 2021-06-17:
On the bottom, you have the Qwiic connector in the wrong orientation.
It should be a Right Angle part. With the opening toward the edge of the board.
Here are the production photos. Those links above don't work

[image]
[image]

eyuan-creare, 2021-06-17:
So this is not the correct part?<br>

[image]

eyuan-creare, 2021-06-17:
I'm going to leave this issue open so I can post progress of the model.

eyuan-creare, 2021-06-17:
Here is a general shape I am heading towards.  I thought it might be nice to taper on the opposite end of the SD card.  Let me know what you think.

[image]

[image]

I know the SD card access is still tight.  I'll work on that next.  I plan on pushing the wall in to provide a nice indentation where your fingers can grab the card.

biomurph, 2021-06-17:
Oh dang, I wanted to get the Right Angle Qwiic connector, but I got the Vertical instead!

biomurph, 2021-06-17:
I have confirmed that the distance between the Tympand E PCB and the Teensy 4.1 PCB is actually 0.165"
Also, the board edges on the USB end line up perfectly.

And yes, I did get the production Rev E delivered today!

eyuan-creare, 2021-06-17:
@biomurph Thanks for taking measurements.  What do you mean by "board edges on the USB end line up perfectly"?

- The RA Quiic would have been nice, but it's a bonus feature.  Are you saying they came with vertical connectors?

biomurph, 2021-06-17:
@eyuan-creare 
When I say that the board edges line up perfectly, I mean that the edge of the Teensy board that the USB is on lines up right with the edge of the Tympan board below it. They are in alignment.

- NOTE the Qwiic connector placed is not the Qwiic connector that I requested in my BOM.

**Please proceed under the assumption that we are using the correct, RA Qwiic connector.**
I am working with Circuit Hub to address the issue.

eyuan-creare, 2021-06-17:
### SD Slot Access
- Let me know what you think of this SD slot.  @biomurph If you are up to printing a preliminary shape, I'll put rudimentary holes for all the connectors then send you a model that's just split in half.  Then we could check how hard it is to remove the SD card.

[image]

[image]

[image]

[image]

biomurph, 2021-06-17:
This looks great!
Yes, please upload STL or OBJ files and I will do a test print to see how the parts fit and the holes line up.

eyuan-creare, 2021-06-17:
So you are saying the edges of the PCB (on the side the USB is on) are aligned: <br>
[image]

biomurph, 2021-06-18:
@eyuan-creare 
Yes. That's exactly it.
I think the board to board spacing is likely variable.
There is some room on the pins to shove the plastic spacer closer to the Tympan, but it's tricky and a pain. Best to leave them and call that dimension 0.165"

chipaudette, 2021-06-18:
@eyuan-creare , be aware that the way that you've drawn the battery wires requires that you bend the wires back on themselves in a way that goes against the way that they're soldered.

The batteries in my hand have their wires soldered on so that the wires extend lengthwise off the end battery.  In your drawing, the wires are bend back to pass over the battery itself.  As a result, we have to bend the wires back against the way that they naturally want to lie.  When I do that bend on the battery in my hand, it all happens where the wire is soldered to the battery.  This will stress the solder points quite a bit.  

I'm sure that we can live with it, but we should be aware of it.

chipaudette, 2021-06-18:
@eyuan-creare , I don't know where you're going to split the case, but it does look tricky to simultaneously slide the PCB into the headphone-end of the case while also having female headers poke up through the holes in the top of the case.  You've made this work in the past, so I have confidence that you're on it...but hooo-weee, you're setting yourself up for a tricky design task!

eyuan-creare, 2021-06-18:
Re: Batteries
Good point.  I was going off of how the 1200mAH battery is packaged.  I'll check how the new ones I ordered look.
[image]

eyuan-creare, 2021-06-18:
Re: Splitting the case
The items that need split are the audio jacks, the P/S & pot, and the USB jack. I could probably tilt the PCB in to omit the split at the audio jacks.  
<br> 
[image]

I'm assuming the SD card must be removed before installing the Tympan.  I always forget this step and stress out the SD slot.  There's a possibility I could split the case at the SD slot.  But that would take some wrangling.  The bottom case would have to slide over the USB slot. @chipaudette let me know if this is worth pursuing:<br>
[image]

chipaudette, 2021-06-18:
I think that no one will remember to remove the SD card from the slot prior to removal, so it's great that you're considering that scenario.  But...

In the RevE, the SD slot is built-into the T4.  The T4 slot is *much* more robust the slot on the RevD.  So, the RevE will be more tolerant to failing to remove the SD.  I think that we'd be fine without having to do the split at the SD slot.

So, I don't feel the need for you to pursue the alternative concept unless you yourself liked it better for a bunch of other reasons.

biomurph, 2021-06-18:
As far as printability, the first design, with the part on the audio jack might be easier.
Also, parting on the jacks and the USB makes it easy to remove the board by lifting the USB end, thus 'sliding' any SD card out.

@eyuan-creare 
Let's print both!
I've got the filament, and I'm ready to go! 
That way we can have some empirical evidence.

eyuan-creare, 2021-06-21:
### Enclosure for Test Print
- Will export these parts for a test print on Monday.  And see if I can add some height to accommodate the new batteries @biomurph found (10.3 x 34.5 x 52 mm).<br>
[image]

[image]

Perhaps I was influenced by my 5yr-old, who just discovered eyebrows :)
[image]

[image]

[image]

[image]

Should I punch a hole in the case for the Quiic connector?  Hoping I don't have to as it would leave an ugly gap in the case.

[image]

biomurph, 2021-06-21:
@eyuan-creare 
That is really great looking work!
I will be shipping with the Adafruit batteries in the near term so no need to change the vertical dimension in that regard. Also, I am removing the Teensy program push button to avoid any chance of random resets. Don't bother putting in the quick connector hole yet. Let's get the prints made and see how it fits first

chipaudette, 2021-06-21:
Looks pretty good!

How are the bottom and top held to each other?

eyuan-creare, 2021-06-21:
Right now, by magic SW-glue. 😊

I intend to bolt the PCB Assy to the lid.  I will try to use as few screws as possible, but I may need separate screws for the PCB Assy.  

Going to correct a few features then export it for a test print to see if the parts fit.  Then work on holes for mics and LEDs, then add screw bosses, then a lip.

eyuan-creare, 2021-06-21:
@chipaudette How should I handle the microphone port?  Having the port so narrow can create high-frequency resonance.  However the header and audio jacks are close by, so it would be difficult to drop the height of the case.  

[image]

Should I port the mics out the side?
[image]

Also, should I look into a gasket to seal the mic port?  Or keep the design simple?

chipaudette, 2021-06-21:
..and adding "tympan" on it somewhere, like your previous builds?   :)

...and possibly saying "RevE" somewhere?  Maybe?  Maybe not?

...and consider adding a flat smooth spot for a sticker for the BT ID, like you've done on some of the earlier units.  That's always a super-helpful piece of info to have written on the case.

...and no need to be stingy with the size of the holes for the PCB mics.  Big holes is good acoustically.  Also, drawing attention to the locations of the mics is a benefit, not a curse.   I do agree, however, that aesthetics are important, too...and big holes might not be very attractive.  It's your call how big/small to make the mic holes; I just wanted it clear that I'm not specifically asking for them to be tiny.

biomurph, 2021-06-21:
Are the on-board mics used that much for 'real' audio work?
I seem to remember that @chipaudette said once they are only used for proof testing and rarely used for any real recording?

Super psyched to print this thing!

eyuan-creare, 2021-06-21:
@chipaudette So narrow mic ports are okay, and no need to side port them? I'll shoot for Ø0.125"~0.15"

biomurph, 2021-06-21:
Just got notification that the Tympan Rev E.2 arrived at your location

chipaudette, 2021-06-21:
I think that the on-board mics are most useful to get people to wrap their minds around what this thing can do.  They're good for quick demos, for sure.

For proper recordings, yeah, the weird acoustics of being inside an enclosure is not great.  People would bring their own external microphone for their proper work.  But, like me, I bet they'll end up using the PCB mics a lot more than they'd like to admit...they're there and they're easy to use.  It's hard to argue with that.

biomurph, 2021-06-21:
I kind of like that two-tone blue top and bottom...
I have a couple different colors of blue filament here. Might come out nice!

eyuan-creare, 2021-06-21:
@biomurph Here are STEP and STL files for the test print.  (Note that there are no holes for screws or a lip for the split case parts.) I will update these files as I add new features.

https://github.com/Tympan/Tympan_Rev_E_Hardware/tree/master/Enclosure/Exports

I left Ø0.150" for the mic ports:<br> 
[image]   [image]

eyuan-creare, 2021-06-21:
@biomurph : The Tympan mounting holes are Ø3mm.  #4 screws are too tight a fit, so I think I will go 2.5mm machine screws.  okay?

chipaudette, 2021-06-21:
Thanks for adding the mic holes.

Based on looking at the CAD rendering, I think that you've added tubes going down to the mics, rather than just adding holes in the top surface.  Fancy!  While this seems like a good idea (and it could be!), there's always the question of what happens to the acoustics.  By having the tubes, it might end up making it sound like, you know, that you're listening at the end of a tube.  

The other option isn't great, either.  If you just have holes in the top of the enclosure, you end up hearing the resonance of the whole inside of the enclosure.  That's known to be ugly, too.  Ick.   But, we also know that laying a piece of foam in the enclosure helps cut down on the enclosure's resonance, which maybe makes it better than listening via a tube

Or not.

One really doesn't know until one listens through a prototype.  So, let's print and listen!

biomurph, 2021-06-21:
@eyuan-creare I will print the models now.
When I imported them into my slicer software (Cura, Lulzbot Edition) the size of them was super duper tiny.
I think maybe it's an inch - mm problem? 
I opened the STEP files in FreeCAD and was able to export them as STLs and they do appear to be the right size.
Print will take 2.5 hours to do 1 top and 1 bottom.

@chipaudette the Rev D enclosure does not have 'tubes'. I lay in a piece of foam during assembly.

eyuan-creare, 2021-06-21:
Good to know.  I am about to add screw bosses that could interfere with inserting the battery.  Will update the CAD around 3pm.

biomurph, 2021-06-21:
going with 2.5mm machine screws is a fine idea!

eyuan-creare, 2021-06-21:
I generally try to use plastic self-retaining washers, so that the screws don't fall out of the case when opening the enclosure.  But we are short on height so I think I will leave them out.  I could create a bump out for them if we really wanted to include them.

[image]

eyuan-creare, 2021-06-21:
@biomurph I'll order some heat-set inserts and screws to send to you:
- https://www.mcmaster.com/94180A321/
- https://www.mcmaster.com/90258A121/

biomurph, 2021-06-21:
Thanks for ordering the parts! 
Please place the order today so that they arrive tomorrow.
I have orders coming in for Typman Rev E already!

Here's a link to a [google drive](https://drive.google.com/drive/folders/1FYHjpcABMl_MS20OsRIkHAHM8Abl21QO?usp=sharing) folder with photos of the files I printed today. Let me know if you can't see the photos.
The fit is real nice. I don't think you need to make any adjustment to it. 

It just needs lips and bosses for the screws

eyuan-creare, 2021-06-21:
Hah! That is so cool to see a print of the design in process.  Thanks for running those through.  Will let you know when I update the CAD.  Likely tomorrow morning.

eyuan-creare, 2021-06-22:
@chipaudette Do we need a place to hook a neck harness?  If so I can add one below the SD card, but it might be unbalanced.  Also, do we need rubber feet?

[image]

eyuan-creare, 2021-06-22:
The (2) screws near the audio jacks sandwich the PCBs between the top and bottom case.  But due to how the battery is inserted, I'll need a clip for the opposite side.

[image]

Screws in the top case look a bit ugly, but it's the most direct way of fastening them.  I could use pegs through the PCBs and put the screws on the perimeter of the case, but that sounds complicated and widens the case.

eyuan-creare, 2021-06-22:
Actually, I think there is room to screw from the bottom if I push the case in a little.  That will clean up the top considerably.

[image]
[image]

chipaudette, 2021-06-22:
I don't think that a neck loop is necessary.

[25 more replies]

closed 2021-08-03 by eyuan-creare