Tympan/Tympan_Rev_E_Hardware #16: Preload the BT Module

eyuan-creare, 2021-06-09:
Define how you will be preloading the BT module...both the firmware level (V5?) and the BT settings and BT name... we're in BLE mode now, so this is really important

chipaudette, 2021-06-10:
My perspective:

  * Firmware: The firmware in the near term (ie, any ASA deliveries) must be V5.5, as described and saved in our [Tympan BC127 repo.](https://github.com/Tympan/BC127_Firmware)

  * Configuration:  After ensuring the correct firmware is present, the BLE comms with the Tympan app might be functional, but I'm not sure.  It might be necessary to run some sort of updated version of the [Tympan_Rename ](https://github.com/Tympan/Tympan_Library/tree/develop/examples/02-Utility/RenameTympanBT)sketch.  The specific code in this sketch needs to be nailed down.  The primary concern is that we now leave the BC127 in its COMMAND mode, which is the opposite of what we used to do (we used to leave it in DATA mode)
  
  * BT Name: We used to set the BT name as Tympan XXXX or whatever and everyone was happy.  But, with the switch to BLE, the phone usually displays not the full bluetooth name but some shorter BLE-specific name.  The BLE name is only seven characters, which is super-duper annoying.  I don't know what to do for our convention.  Maybe "TYMXXXX"?  From the BC127 V5 docs, here are the two name settings:

    NAME=value
    Friendly Name; Last 6 symbols of the name XXXXXX are the last 6 symbols of the Bluetooth address.  Example:  BlueCreation-XXXXXX
    
    NAME_SHORT=value 
    Short Name, used for Melody Smart Advertisements, maximum of 7 characters, where XXXXX are the last 5 hex digits of the Bluetooth address.  Example: BCXXXXX
`

biomurph, 2021-06-10:
From what I understand, we will need to make the change to two names.
One is the `NAME` which we already do. We'll call these ones `Tympan E-XXXX` as we have?

The other is `NAME_SHORT` for this, could we just use `TYMXXXX` that way we preserve the unique id?

chipaudette, 2021-06-10:
@biomurph, I believe that the naming convention that you're suggesting is exactly what the latest rename sketch (as linked in my earlier comment) does.  Similar thinking!