Tympan/Tympan_Rev_E_Hardware #18: Expose SD Card

eyuan-creare, 2021-06-09:
### Options
1. Bump in the case and move the battery back

[image]

2. Place a 18650 to the side

[image]

chipaudette, 2021-06-09:
Are there easily-obtained cylindrical cells that are narrower than the 18650?  Like, are 14500 readily available?  https://en.wikipedia.org/wiki/List_of_battery_sizes#Cylindrical_lithium-ion_rechargeable_battery

eyuan-creare, 2021-06-09:
Fenix is a popular brand among flashlight enthusiasts, so I believe it's 1600mAh rating .  However it is unlikely we will find an off the shelf battery with **leads**.  I just don't like widening the case, even if it is less overall volume.

https://www.batteryjunction.com/fenix-arb-l14-1600u.html

biomurph, 2021-06-10:
We have two issue threads covering this problem.
@eyuan-creare @chipaudette 
I have put candidates on the other thread for flat pack and cylinder battery options to review.

eyuan-creare, 2021-06-10:
@chipaudette, @biomurph I'd like to rule out a battery on the side.  I think it makes the case too awkward to put in your pocket and prefer making the case jog in to expose the SD card.  Do you agree?  Should we move forward with option 1?

biomurph, 2021-06-10:
I would also prefer not to have the battery on the side.
If one of the candidates from the other thread will work, and fit to allow access to the SD card, then awesome!

chipaudette, 2021-06-10:
The jog in the case is not my favorite (it makes my mind think about our not-so-attractive RevC case) but if everyone else has reservations with the side battery, I'm not opposed to the job in the case.

I support the job in the case enough to take the next few steps in the design process: (1) pick a candidate battery, (2) insert the battery in the model, (3) conceptually decide where our fasteners will go, (4) illustrate the new SD slot and so that we see how easy/hard it might be to access the card.

chipaudette, 2021-06-10:
I do have concerns over the difficulty of getting the card in and out (hency my request for bullet #3 above).  As currently drawn, the jogged wall really blocks your fingers from holding the SD card as you try to slide it in/out.

[image]

That's the kind of access issue, though, that I'm looking for ideas on.

eyuan-creare, 2021-06-10:
Also, it can be difficult to guide the card into the slot without some support.  And adding support to the case can be difficult depending on whether PCBs need to slip past the support to be installed.

Here are some options for exposing the card:

1. We let the SD card stick out a bit and create an indentation for your fingernails or tweezers to get in there

2. We make a false cover that clips in. If we are feeling adventurous, we could make it flip out on a living hinge.  That would retain the original look somewhat but is a bit of a design effort and limits material choices.

[image]

PDF to scale: 
[case_ideas.pdf](https://github.com/Tympan/Tympan_Rev_E_Hardware/files/6632964/case_ideas.pdf)

@biomurph What kind of materials can you print, and what colors can you do?

chipaudette, 2021-06-10:
If you're proposing the 2000 mA-hr, where do the fasteners down there fit?  Or are you saying we don't need fasteners down there?

eyuan-creare, 2021-06-10:
@chipaudette: Both the Adafruit 2000mAh and 1200mAh are 2.4" in length.

For option 1, If we need to we can lengthen the case to clear the battery wires and make a large indention for the USB plug.  It will take some effort to mock up so we should choose which design you like best (given the risk of more complex designs) and run with it till it breaks.

biomurph, 2021-06-10:
I want to make this happen with as little adventure as possible. The case print files should be easy for anyone to print on their own.

I have a Tax mini 2, and it rocks PLA all day! Any color

biomurph, 2021-06-10:
Also, @eyuan-creare you're only showing the side view. We may get a better sense of access with a view from the short end?

I don't mind a cut away that is kind of like the Rev C, personally. But again, if it can be easy (in quotes).

The lipos that I have listed in the other thread are preferred because they are accessible and affordable.please use dima from those only

eyuan-creare, 2021-06-10:
Here is another option, though I am doubtful that we can source a battery narrow enough to fit between the headers.  We would likely drop in capacity quite a bit to make this happen.

[image]

eyuan-creare, 2021-06-10:
Here's a view of the existing case looking into the audio jacks.  Pink is the SD card.  I also updated the scale drawing.
[2021-0610-01_case_ideas.pdf](https://github.com/Tympan/Tympan_Rev_E_Hardware/files/6633821/2021-0610-01_case_ideas.pdf)

[image]

eyuan-creare, 2021-06-11:
@chipaudette You asked about locating screws.  I had to do a workaround for the last revision too, where the heat set inserts are above the battery.

[image]

For the single screw near the USB connector, we would need to do something similar.

[image]

Since Joel will be printing them, it gives us a chance to do more iterations.  We could consider a snap fit case, depending on how elastic the PLA material is.

biomurph, 2021-06-11:
I have found that I get a nice tight for with the forms from RevD. If you can look at those models in the Rev D repo, the way the cases fit comes out good on my machine.

I would avoid a fit that relies on the plastic unless it is robust clip on one end and screw on the other? Snap fit becomes break fit after a while.

I do want to use the threaded inserts on this build. They are nice

biomurph, 2021-06-11:
I've just looked around and it seems that PET or PETG would be a good material instead of the PLA.
I think that for the near term, I will use PLA because I have yield from my printer, and then transition to the PETG when there's time to tweak and twiddle the print settings. I have read that it will print as easily as PLA, but getting there can be a bit of a climb. Also, there may be issues with support removal and 'stringing', that is, stringy strings around the part left during print head moves. All of this sounds like it can be overcome and prints will come out good, but for our close deadlines I want to use what I know works. 

When I get a more UV resistant material, I will can send out replacement cases

eyuan-creare, 2021-06-11:
Perhaps this is a combination of practical and shapely.  There would only be two screws to access the SD card.  It's possible we could make the bottom half snap fit onto the top, with screw holes in case you wanted it to be more secure.

@chipaudette, @biomurph  if you have time today, perhaps we should all chat.  My schedule is pretty free.

[image]

biomurph, 2021-06-11:
I will be able to be on the phone later today.

I think that the idea of a 'hook' or 'tab-slot' type capture on the USB end, while having a screw on the jack end could be nice.

Are you now thinking that we *won't* be putting a slot for the SD and that it will be accessed via case removal?
This would be good *if* we can get the SD card playback working, is that correct?

eyuan-creare, 2021-06-11:
The idea behind (7) is that the bottom would be easily removeable, by snap fit to the top or perhaps sliding in, or screws.  That way a user could easily access the SD card or swap a battery. It will take some geometry acrobatics to make it happen.  If I have trouble, then I could revert to a snap in cap (3) which would be a 3pc design. 

If we wanted the SD card more accessible, then I think a bump out like (1) or (4) would work.

biomurph, 2021-06-11:
OK!

I am all for whatever is going to be the most easiest to model, print, assemble, etc.
If it's a removable bottom to get the SD card out and option to swap battery, that's kinda cool!

I have had some luck myslef with a 'one end latch, the other end screw' design. That might be the easiest.

chipaudette, 2021-06-11:
SD Playback (if you're referring to the bug identified by our friends in Montreal) is unrelated to the need to physically access the SD card.  They are separate issues.

What would relieve our need for physical access is if the MTP functionality were to suddenly be much more mature and if it were part of the Teensyduino install.  MTP lets you electronically access the SD card through the Tympan from your PC.  It basically turns your Tympan into an SD reader such that you never need to physically remove the SD.  

Unfortunately, MTP is neither mature enough nor part of the Teensyduino installation, so having physical access to the SD still seems important.

chipaudette, 2021-06-11:
@eyuan-creare , I am open to using screws (concept 7).  Looking back at Concept 1, I see that your rendering of Concept 4 actually addresses most of my concerns about getting your fingers / tweezers onto the SD card.  

[image]

The part of Concept 4 that is the crucial difference from Concept 1 is that you have raised that upper wall to get it a bit more out of the way of the SD slot.  To me, that seems like enough of a shift to make it workable, even without the SD holder.

What do y'all think of modifying Concept 1 to have the higher wall of Concept 4 (but otherwise staying as Concept 1)??

If it were to work for you, then we'd "just" have to figure out how to do the fasteners (and/or hook) and how to make room for the lead wires to come off the battery.

eyuan-creare, 2021-06-11:
Yes, there shouldn't be a problem raising that wall.  That 3-pin header between the audio jacks shouldn't be there in a production unit.  Though we may want to leave a wall so that the SD card slides against something.  The tip of the SD card has an edge to it that could allow a screwdriver or pick to flick it out.

eyuan-creare, 2021-06-17:
Closing this, as we decided on an approach.  I'll follow up with updates on #15

closed 2021-06-17 by eyuan-creare