Tympan/Tympan_Rev_E_Hardware #19: Preinstall an SD card?

chipaudette, 2021-06-10:
Depending on what we decide for the enclosure, do we preinstall an SD card in the ASA units?

biomurph, 2021-06-10:
Yes, depending.
I would prefer not to have to add an SD card to the BOM, but it would ensure that they get a good, fast card.

Would [these](https://www.amazon.com/JUANWE-High-Speed-Smartphone-Bluetooth-Speaker/dp/B086DKQQ6F/ref=sr_1_6_sspa?dchild=1&keywords=micro%2BSD%2Bcard&qid=1623332213&s=toys-and-games&sr=1-6-spons&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUE3QThMS0hTMTJDV0EmZW5jcnlwdGVkSWQ9QTA1ODI0MTlNODFFWENFT0Y2VkEmZW5jcnlwdGVkQWRJZD1BMDU0NzIxNzM2QU02MEhVMEpUQTUmd2lkZ2V0TmFtZT1zcF9idGYmYWN0aW9uPWNsaWNrUmVkaXJlY3QmZG9Ob3RMb2dDbGljaz10cnVl&th=1) cards be a good candidate? They don't seem to have any speed certification, but they are cheap... you do get what you pay for...

chipaudette, 2021-06-10:
For recording audio, the Tympan RevD has been *extremely* sensitive to the card that one uses.  I can't speak to SD on the RevE.

For the RevD, we've used [SanDisk Ultra 32GB](https://www.amazon.com/SanDisk-Ultra-microSDXC-Memory-Adapter/dp/B073JWXGNT/ref=sr_1_3?dchild=1&keywords=Sandisk+ultra+32GB&qid=1623332805&sr=8-3).  Chris found these.  None of the other SD cards gave nearly as good performance.

Recently, we had a hard time getting the Ultra, so we bought some [SanDisk Extreme 32GB](https://www.amazon.com/SanDisk-Extreme-Memory-Adapte-SDSDQXN-032G-G46A-Version/dp/B00M55BS8G/ref=sr_1_1_sspa?dchild=1&keywords=Sandisk+extreme+32GB&qid=1623332933&sr=8-1-spons&psc=1&smid=A206L2ND1MNB5K&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEyVlo3VDE4V1JRRzdOJmVuY3J5cHRlZElkPUEwMDE1NjgwM1IzQUtKMEpMWEJVVyZlbmNyeXB0ZWRBZElkPUEwNjUxODk0SkNGRFJBNUwyODhIJndpZGdldE5hbWU9c3BfYXRmJmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ==).  We don't know if they perform as well as the Ultra.  Anecdotally, Chris has not had quite as good an experience, but the underlying SD library has been changing, too.  So it's hard to know.

We should not use the cheapest cards as everyone will just be disappointed with the inability to write audio without hiccups showing up every few seconds. 

I'm not proposing that we pre-install SD cards long-term.  I'm only proposing that we do it for the ASA units.  I vote that we use the best cards that we know about.

biomurph, 2021-06-10:
@chipaudette I agree that we should use the best, and that the low cost ones can be glitchy.
In my experience, I use lower level mcu that do not run native SD (they use and SPI hack). Those systems definitely need high quality SD cards. Now I know that the Teensy (and Tympan) do as well.

It would be a nice perk to add the SD card to the ASA kit. I will look for the ScanDisk Ultra.

biomurph, 2021-06-10:
I've sourced the ScanDisk Ultra 32GB

chipaudette, 2021-06-10:
Thanks Joel.  When I said  "We can buy them", I meant "Creare can buy them".  Upon re-reading, that wasn't clear.  Sorry.  If you would prefer that Creare buy them, let me know.

biomurph, 2021-06-10:
I'll email you a link to the cards.