Tympan/Tympan_Rev_E_Hardware #25: BC127 Module Variant Availabiltiy

biomurph, 2022-02-14:
@chipaudette @eyuan-creare 

I have Tympan Rev E boards prototyped to test the only available BC127 modules.
The two modules that are available are:

- [BC127-EXT-HD](https://www.digikey.com/en/products/detail/sierra-wireless/BC127-EXT-HD-1104314/11657575)
- [BC127-EXT-APTX](https://www.digikey.com/en/products/detail/sierra-wireless/BC127-EXT-APTX-1103783/7801750)

The current Melody Firmware that we are using is the `Melody 7.3-HD`.
I have uploaded the latest Melody Firmware v7.3 to the modules under test, but neither of them would take the `Melody 7.3-HD` which we are currently using. They only take the `Melody 7.3-STANDARD` firmware. 

I don't know what the difference is between these firmware variants, but I am hoping that it is negligible or not applicable to our application. 

I need to solder in the through-hole components to these boards and then run tests with the Tympan Remote app.

Do you have any suggestions for the best app example for testing? I am hoping that there is a GO / NO GO test that can be run to verify these modules with the STANDARD firmware. If I get the basic gain app running and connected to the Tympan under test and the behavior is as expected, is that enough to call it good?

I will ship one of these to you so that you have one in house that you can test with if we run into a jam.

eyuan-creare, 2022-02-14:
Thanks for your due diligence on this.  In case it helps, I just received an order from Digikey for standard BC127 modules, however, the order was placed 5months ago.

biomurph, 2022-02-15:
How many did you order?

chipaudette, 2022-02-15:
@biomurph, the `Melody 7.3-HD` firmware ought to work, but only with the BC127-EXT-HD version of the module.  For the BC127-EXT-APTX module, I'm guessing that only the `Melody 7.3-STANDARD` would work (unless there were to be some sort of `Melody 7.3-APTX` version available from Sierra).

chipaudette, 2022-02-15:
As for testing / debugging, I don't think that I have anything as simple as go/no-go...but if you (1)  get the Basic Gain sketch running on the tympan, (2) you can connect from the phone App, and (3) you can make changes from the phone and have the Tympan sound different (ie, get louder), you win!

Another test would be to run this example sketch: https://github.com/Tympan/Tympan_Library/blob/main/examples/07-Bluetooth/BasicGain_wApp_BLEdebug/BasicGain_wApp_BLEdebug.ino

In that sketch, open the Serial monitor and restart the Tympan.  Watch all the debugging messages as it tries to talk to the Bluetooth module.  If you get sensible replies (as opposed to ERRORS, or as opposed to no replies at all), you're probably good.  Also, you can try connecting from your phone and this sketch will spew more debugging info to the Serial Monitor during that connection process.

biomurph, 2022-02-15:
@eyuan-creare 
I plan to have on the order of 50+ units made, so the 10 you have are not useful.
If you have a contact info for sales rep, please pass it on. I have had horrible experience in communicating with them.

@chipaudette 
Both of the units, the `BC127-EXT-APTX` and the `BC127-EXT-HD` would only take the `Melody 7.3-STANDARD` firmware. I tried to put the `Melody 7.3-HD` firmware on them and the firmware updater failed to load with an error saying that the hardware is incompatible. So we're running with the `STANDARD` version.

I was able to get the BasicGain_wApp running connected to the Tympan Remote App (Android), and saw no errors in the serial port. All behavior seems normal there.
I will re-test with the BLEdebug code you point to to make sure there are no errors. 
If (when!) there are no errors, I will pack one of the `APTX` Tympan up and ship it to you so that you can have one. I have the case printed in a different color to differentiate it from any other Tympan you might have laying about. Pics coming on this thread.

For clarity and to avoid confusion:

- `EXT` means that the antenna is external, and I need to design antenna circuitry for it
- `HD` means high definition audio. Not sure why they do that...
- `APTX` means you get access to the aptX features. Which ones are included? I don't know, but I'm hoping the [low latency](https://www.aptx.com/aptx-low-latency) is available for us to play with.

chipaudette, 2022-02-15:
The units that you built for ASA used the BC127-HD (not "ext"), right?  Those modules required the HD version of the firmware.  If the new units truly are BC127-EXT-HD, I'm shocked that they do not use the HD firmware.

In the end, I don't really care either way.  We don't use the HD mode, so it's irrelevant to us.  If the system runs, that's good enough for me.

biomurph, 2022-02-15:
@chipaudette 
I'm not too surprised about the HD weirdness. Par for the course with Sierra Wireless.
They 'work' with the debug code. No errors!

I'm shipping one to Creare. Who's attention should it go to?

I printed a green cover so we can tell them apart as prototypes ;)

[image]

biomurph, 2022-02-15:
@eyuan-creare 
I saw your email from the Sierra folks.
Let me know if I can help out with your application.
I have a branch of this repo called [bc127-EXT](https://github.com/Tympan/Tympan_Rev_E_Hardware/tree/bc127-ext) which has the chip antenna circuit and layout on it, which you may be able to use?

chipaudette, 2022-02-15:
@biomurph , Cool.  You can send it to my attention.